ADVRHumanoids/ROSEndEffector #5: Rate of Universal ROSEE should comes from ROS param server

liesrock, 2019-12-10:


torydebra, 2019-12-11:
also the yaml name in
    p.init ( ROSEE::Utils::getPackagePath() + "/configs/two_finger.yaml" );
should be better to have as param in roslaunch file (so we don't have to change the name here when other launch file is launched.
I already did this,  I can do the same for the rate

liesrock, 2019-12-11:
Thanks @torydebra, in the latest master a pushed this morning should be as you described: can you confirm it?

torydebra, 2019-12-11:
Sorry, I did not see that commit.
I had put the param in launch file inside the UniversalRos node
`   <node type="UniversalRosEndEffector" name="UniversalRosEndEffector" pkg="ros_end_effector" output="screen" >
        <param name="yamlConfigFileName" type="str" value="/configs/two_finger.yaml" /> 
    </node>`
 but it is better to have global and take it into the parser as you did

For this issue, isn't easy? like:
```
ROSEE::UniversalRosEndEffectorExecutor::UniversalRosEndEffectorExecutor ( std::string ns ) : _nh ( ns ) {

    if (! _nh.getParam("/rate", _rate)){
        ROS_INFO_STREAM ("Parameter for rate not found, setting default rate of 100");
        _rate = 100.0;
    }
```

liesrock, 2019-12-11:
yes, that's great! Please push it on a branch so we can think about some testing about it and later on if Travis CI gives the ok I will merge it.

liesrock, 2019-12-13:
Merged!

closed 2019-12-13 by liesrock