ADVRHumanoids/ROSEndEffector #16: Passive Tag of srdf should be used in Parser

torydebra, 2020-02-18:
There exist a `<passive_joint\>` tag in srdf that we can use to not consider the joints that are active in the urdf (not fixed nor mimic) but that we want to be "passive" 
Example:
```xml
<passive_joint name="finger_1_joint_2" />
```

liesrock, 2020-04-18:
@torydebra can you take care of this in the Parser?

torydebra, 2020-04-20:
I can, only a clarification.
Now fo example in schunk we have i Parser:
~~~
[ INFO] [1587373162.412185178]: Actuated joints in finger: thumb
[ INFO] [1587373162.412551272]: left_hand_Thumb_Opposition
[ INFO] [1587373162.412730434]: left_hand_Thumb_Flexion
[ INFO] [1587373162.412868420]: left_hand_j3
[ INFO] [1587373162.412998831]: left_hand_j4
[ INFO] [1587373162.419089170]: -------------------------
[ INFO] [1587373162.419692853]: Fingers in EEInterface: 
...
~~~
j3 j4 are passive/mimic. Now, I think they are put in the joint lists (ie maps) along the actuated joints. If you do not send command to these joints, no problem happen. 
What we want is to exclude these joints right? To turn the print into:
~~~
[ INFO] [1587373162.412185178]: Actuated joints in finger: thumb
[ INFO] [1587373162.412551272]: left_hand_Thumb_Opposition
[ INFO] [1587373162.412730434]: left_hand_Thumb_Flexion
[ INFO] [1587373162.419089170]: -------------------------
[ INFO] [1587373162.419692853]: Fingers in EEInterface: 
...
~~~

So, if someone send a command to j3, j4 joint, the code will return a error of joint not found. Correct?

liesrock, 2020-04-20:
Yes this is correct: thanks for the clarification.

So this is linked (and che be solved) with https://github.com/ADVRHumanoids/ROSEndEffector/issues/30#issuecomment-609033244

right?

torydebra, 2020-04-20:
Yes, I was looking what I done in #30 and relative issue_30 branch. A mimic is not considered anymore as actuated. A passive is still not considered, we need another method from the one used for mimic because passive is a info that is in the srdf and not in the urdf

torydebra, 2020-04-20:
I created the pull request. Now in parseSRDF, after filling the maps, passive joint are removed from them.

I also add another map joint->finger, and the relative getter

As discussed, probably the getters for the other map, finger_joint, should be both public. If the case, also another get for the new map should be put in the code (now there is only the one for the copy)

torydebra, 2020-04-20:
We can safely merge into devel; issue30 (that need additional work) will be easy to merge in future

torydebra, 2020-04-21:
Merged, we can close I suppose

closed 2020-04-21 by torydebra