ADVRHumanoids/ROSEndEffector #19: Move msgs and srv to another package

torydebra, 2020-02-26:
Should we create a new rosee_msg package instead having msg and srv folders in the main package?

liesrock, 2020-03-02:
Fully agree with you: I will create a repo and the related dependencies on the main ROSEndEffector package

torydebra, 2020-03-02:
If useful, I already init the package with msg for gui: [https://github.com/torydebra/rosee_msg](https://github.com/torydebra/rosee_msg)

torydebra, 2020-03-05:
I transfer the msg repo from my personal github to ADVRHumanoids, now the new link is https://github.com/ADVRHumanoids/rosee_msg/

It already contains the msg that are also in this repo

liesrock, 2020-03-16:
Thanks @torydebra : the repo is ok!
The only issue now is to adjust the documentation with the new workflow (we need to download also the new repo which is now a dependency) and the Travis config.

I will take care of the Travis, can you update the documentation? If everything is ok I will close this issue later.

torydebra, 2020-03-16:
I already did the doc update: it is in the [devActions](https://github.com/ADVRHumanoids/ROSEndEffector/tree/devActions) branch readme and so included in the pull request #18

torydebra, 2020-03-16:
Very important to remember to select a particular branch when cloning this main repo but also the other "support" ones. This because ofter the master is not the newest one. I add this in readme (f45fa0fa73acf02cf183ad1d85a07327f1a878c6)

liesrock, 2020-03-16:
Thanks @torydebra, we usually use the YCM superbuild to solve the above problem, but we should investigate how to solve it using a ROS-based build system.

liesrock, 2020-03-16:
I fixed with my latest commit (ba4bad8cce9f24d8028ecafc4cb0ac50b0e96e98) also some old msg issues. I am testing the new travis config.

liesrock, 2020-03-16:
OK solved everything: https://travis-ci.org/github/ADVRHumanoids/ROSEndEffector/builds/663169633

torydebra, 2020-03-17:
I reopen it to ask if now it is safe to delete msgs folder in this package

liesrock, 2020-03-17:
Yes, in this commit https://github.com/ADVRHumanoids/ROSEndEffector/commit/ba4bad8cce9f24d8028ecafc4cb0ac50b0e96e98

If I forgot to delete the msgs folder please do it.

torydebra, 2020-03-17:
done 30794fcad6325a98727116851b8ed02064913e62

closed 2020-03-17 by torydebra