ADVRHumanoids/ROSEndEffector #22: Joint with id == 0 not allowed to move in pinch callback

torydebra, 2020-03-04:
In UniversalRosEndEffector.cpp, we have  function 
`void ROSEE::UniversalRosEndEffectorExecutor::pinchCallback ( const ros_end_effector::EEPinchControlConstPtr& msg )` where we have:

~~~c++
 for ( auto it : pinch_joint_involved_mask ) {

            if ( it.second  != 0 ) {
                int id = -1;
                _ee->getInternalIdForJoint ( it.first, id );
                
                if( id > 0 ) {  //WHY NOT >= ?
                    // NOTE assume single joint
                    _qref[id] = pinch_js.at ( it.first ).at ( 0 ) * msg->percentage;
                }
                else {
                    ROS_WARN_STREAM ( "Trying to move Joint: " << it.first << " with ID: " << id );
                }
            }
        }
~~~

Why a joint with `id == 0` is not allowed to move? 
0 is a valid id given to the first joint encountered while parsing the model... am I missing something?

I found this while trying to play with robotiq_3f hand, where for a pinch I need to move the first joint, (that has id ==0) but a ROS_WARN is given: 
~~~
[ WARN] [1583322108.722824700]: Trying to move Joint: palm_finger_1_joint with ID: 0
~~~

liesrock, 2020-03-04:
Thanks a lot @torydebra to point this out: yes it is a bug, I will fix it asap.

liesrock, 2020-03-16:
Solved by @torydebra in e7a4818f : I will merge it in devel asap.

closed 2020-03-16 by liesrock