ADVRHumanoids/ROSEndEffector #24: Question about equations in Utils.h

gtan039, 2020-03-12:
Hi, I have been looking through the different libraries and scripts and I have a question about some of the equations used, namely in `Utils.h`, in the functions `process `and `computeCoeff`

for `process`, what do the variables mean and what is the concept behind deriving `_y`? 

        _y = 1.0/_a0 * ( _u + _b1*_ud + _b2*_udd - _a1*_yd - _a2*_ydd );

for `computeCoeff`, similarly what is the mathematical concept behind deriving of `_a0`,`_a1` and `_a2`?

    _a0 = 1.0 + 4.0*_eps/ ( _omega*_ts ) + 4.0/std::pow ( _omega*_ts, 2.0 );
    _a1 = 2 - 8.0/std::pow ( _omega*_ts, 2.0 );
    _a2 = 1.0 + 4.0/std::pow ( _omega*_ts, 2.0 ) - 4.0*_eps/ ( _omega*_ts 

Thank you so much for your help and clarification

liesrock, 2020-03-12:
Hi @gtan039, this is a utility class for a second order filter implemented by my colleague @alaurenzi.

alaurenzi, 2020-03-12:
I have added some doxygen description in the XBotInterface repo:

```
/**
 * @brief SecondOrderFilter implements a canonical continuous-time
 * second order filter with transfer function
 *                   1
 * P(s) =  -----------------------,  w = natural frequency, eps = damping ratio
 *         (s/w)^2 + 2*eps/w*s + 1
 *
 * and discretized according to a trapezoidal (aka Tustin) scheme. This yields
 * a difference equation of the following form:
 *
 *      a0*y + a1*yd + a2*ydd = u + b1*ud + b2*udd
 *
 * where yd = y(k-1), ydd = y(k-2) and so on (d = delayed).
 */
```

@liesrock maybe you can add it here as well

liesrock, 2020-03-12:
Thanks a lot @alaurenzi, I will integrate it properly.

@gtan039 I hope this is clear now.

closed 2020-03-12 by liesrock