ADVRHumanoids/ROSEndEffector #25: [Tests] FindPinch and FindTrigs fail

liesrock, 2020-03-16:
Currently the two above tests fail: you can reproduce it by cloning on a new catkin the ROSEndEffector and the ros_ee_msg repos.

Investigating the code I notice that for example in test_find_pinches.cpp

```
        auto theTwoMaps = actionsFinder.findPinch(folderForActions + "/primitives/");
        pinchMap = theTwoMaps.first;
        pinchWeakMap = theTwoMaps.second;

```

I noticed that the primitives folder does not exist so we have a Segfault.

@torydebra Can you check both the tests or push the missing files?

torydebra, 2020-03-16:
So strange.
It is correct that the primitives folder does not exist before running the tests, because the code still have to create the action yaml files and the relative subfolders. 
That snippet you posted is equal to the one in the non-test code [UniversalFindActions](https://github.com/ADVRHumanoids/ROSEndEffector/blob/devActions/src/UniversalFindActions.cpp) : 
~~~ c++ 
std::string folderForActions = ROSEE::Utils::getPackagePath() + "/configs/actions/" + parserMoveIt->getHandName();
    
ROSEE::FindActions actionsFinder (parserMoveIt);

auto maps = actionsFinder.findPinch(folderForActions + "/primitives/");
~~~

In fact, both test and UniversalFindActions run well in local on my pc (I do not try with completely new repo, but I deleted the configs/actions folder and when running the code all the subfolders are correctly created).
The problem you have is on your pc or in travis server?

Anyway, the folders are created when FindActions call the function (after it has looked for for a specific primitive): 
~~~c++
yamlWorker.createYamlFile(mapForWorker, "pinchStrong", path2saveYaml);
~~~
The yamlworker, before creating the file, create the folder (and subfolders):
~~~c++
std::string ROSEE::YamlWorker::createYamlFile(
    
    const std::map < std::set <std::string> , ActionPrimitive* > mapOfActions,
    const std::string actionName, std::string pathFolder) {
    
    ROSEE::Utils::create_directory ( pathFolder );
    std::string output = emitYaml ( mapOfActions );
    ROSEE::Utils::out2file( pathFolder + actionName + ".yaml", output);
    return (pathFolder + actionName + ".yaml");
    
    
}
~~~
`ROSEE::Utils::create_directory ( pathFolder )` uses boost to create pathFolder folder and subfolders if these do not exists, otherwise it does nothing.

So I do not know where the error is. Does `ROSEE::Utils::create_directory ( pathFolder )` act correctly on your machine? Maybe some permission issues?

My other question is: should I move `ROSEE::Utils::create_directory ( pathFolder )` in UniversalFindActions.cpp for better readability?

torydebra, 2020-03-16:
Forgot to say, I missed a "/" character in the pathfolder of test. I added it in the last commit, anyway the code for me worked also before this correction.

liesrock, 2020-03-16:
So the issue is both for my PC and for Travis: I will check your comments and commits and come back to you here.

liesrock, 2020-03-16:
Looks like the issue is not on the directories (as you pointed out), but on the MoveIt parser initialization: 

```cpp
std::shared_ptr <ROSEE::ParserMoveIt> parserMoveIt = std::make_shared <ROSEE::ParserMoveIt> ();
parserMoveIt->init ("robot_description") ;
ROSEE::FindActions actionsFinder (parserMoveIt);
```
This leads on the following errors on the command line:

```
[ERROR] [1584370767.560095567]: Robot model parameter not found! Did you remap 'robot_description'?
 [PARSER::init]: Fail To load robot model robot_description

```

and running gdb I arrived to this line:

```cpp
    planning_scene::PlanningScene planning_scene ( parserMoveIt->getRobotModel() );
```

on FindActions.cpp

Any ideas?

liesrock, 2020-03-16:
On test_find_trigs.cpp instead the error is related with **checkNameTypeConsistency** test, in particular the fails appears at this call:

```
actionType = trigParsedMap.at(k).begin()->second->getPrimitiveType(); 
```

which is line 125.

torydebra, 2020-03-16:
Fast answer then I will check better the errors:
Are you launching the find actions with findActionsSchunk.launch? 
If using the schunk, have you sourced the schunk package before launching? (It is necessary because schunk models are taken from their package)

liesrock, 2020-03-16:
No everything is running from simple executable on command line (both on travis and my pc). No worries take your time.

torydebra, 2020-03-16:
So, maybe the problem is that you do not set the urdf and srdf files on ros parameter server before running the node. 
Both on .travis:
~~~yaml
#test with test_ee hand
- rosparam set -t $(rospack find ros_end_effector)/configs/urdf/test_ee.urdf robot_description
- rosparam set -t $(rospack find ros_end_effector)/configs/srdf/test_ee.srdf robot_description_semantic
- make test ARGS="-V"
~~~
and in launch (eg for the schunk)
~~~xml
  <param name="robot_description" textfile="$(find ros_end_effector)/configs/urdf/two_finger.urdf"/>
  <param name="robot_description_semantic" textfile="$(find ros_end_effector)/configs/srdf/two_finger.srdf" /> 

 <node type="EEInterface_test" name="EEInterface_test" pkg="ros_end_effector" output="screen"/> 
<node type="FindPinches_test" name="FindPinches_test" pkg="ros_end_effector" output="screen"/>     
<node type="FindTrigs_test" name="FindTrigs_test" pkg="ros_end_effector" output="screen"/>
<node type="ComposedAction_test" name="ComposedAction_test" pkg="ros_end_effector" output="screen"/>
~~~

Moveit parse the model from ros param server through its internal functions

liesrock, 2020-03-16:
This is correct for my pc: anyhow we should avoid to segfault. Can you add a check on the pointers?
For the travis issue I will investigate.

Thanks!

torydebra, 2020-03-16:
> On test_find_trigs.cpp instead the error is related with **checkNameTypeConsistency** test, in particular the fails appears at this call:
> 
> ```
> actionType = trigParsedMap.at(k).begin()->second->getPrimitiveType(); 
> ```
> 
> which is line 125.

Is this error still present?

> This is correct for my pc: anyhow we should avoid to segfault. Can you add a check on the pointers?
> For the travis issue I will investigate.
> 
> Thanks!

Do you mean a check for the parameter if it is set (this is already implied in the error `[ERROR] [1584370767.560095567]: Robot model parameter not found! Did you remap 'robot_description'?
 [PARSER::init]: Fail To load robot model robot_description` I think.
Or do you mean a check in the tests to see if primitives and other folders exist?

torydebra, 2020-03-16:
> Or do you mean a check in the tests to see if primitives and other folders exist?

Also I think this is not what create the segfault. If the folders does not exist (e.g. we have a hand with no action) the maps are created empty, and simply all the for in the tests will not be executed (because empty maps). 
I tested commenting the findPinch: 
~~~c++
 //auto theTwoMaps = actionsFinder.findPinch(folderForActions + "/primitives/");
std::pair <  std::map < std::pair <std::string, std::string> , ROSEE::ActionPinchStrong >, 
             std::map < std::pair <std::string, std::string> , ROSEE::ActionPinchWeak > > theTwoMaps;
~~~
The result is that the test is passed anyway, there are only lot of prints about file not found (that is correct)
~~~ bash
[ERROR YAMLPARSER:: parseYamlPrimitive]: file /home/tori/ROSEE/src/ROSEndEffector/include/ROSEndEffector/../..//configs/actions/tests/two_finger_end_effector/primitives/pinchStrong.yaml not found. 
[ERROR YAMLPARSER:: parseYamlPrimitive]: file /home/tori/ROSEE/src/ROSEndEffector/include/ROSEndEffector/../..//configs/actions/tests/two_finger_end_effector/primitives/pinchWeak.yaml not found. 

~~~

liesrock, 2020-03-16:
The error is solved for both tests.

The folder is not the issue, as you mentioned, the problem is related to accessing some pointers that are not valid if the parser initialization is not valid.

So the checks I would add are related to the consistency of the parser pointer or on a check related with the initialization which should be with no issues otherwise we can have a potential SEGFAULT.

torydebra, 2020-03-16:
See if this is what you mean : f04b5b2058f9d9521d15dbecd882b9b880d99f15 & 1b77d73e92feead0d6ec3135e6476e9cff3ea0a6.
Now the execution stop if init return false (a bool was already returned but the main did not check it )
In the test I put an ASSERT_TRUE for similar reason