ADVRHumanoids/ROSEndEffector #26: Change Name to MoreTips primitive

torydebra, 2020-03-16:
I do not like so much the name **ActionMoreTips**, for the action that move MORE than one TIPS with a single joint. It is confusing. I want to change it into **ActionSingleJoint** or something like that. 
With this name the user easy remembers that the key for this kind of primitive is a string containing the joint name which moves the more tips. Any better suggestions @liesrock ?

liesrock, 2020-03-26:
Sorry I miss this: I would call it **ActionSingleJointMultipleTips** 
Even if this looks long, it is probably more intuitive to understand for the user.

torydebra, 2020-03-26:
Right, I will change the name to **ActionSingleJointMultipleTips**

torydebra, 2020-03-30:
Done in 6209d4bfd2321f180ce114b2780240f3af8f18f5 . Changes are on rosAction branch and will be included in devel after the pull request #28

closed 2020-03-30 by liesrock