ADVRHumanoids/ROSEndEffector #27: [Build System] Find a proper way to deal with ROSEE sources dependencies

liesrock, 2020-03-17:


liesrock, 2020-04-01:
Currently I am exploring a catkin_workspace based solution.

@torydebra did you find any other mechanism to work on this defining ROS sources dependecies through different repos?

liesrock, 2020-04-01:
Ok currently all devel build ok and are at the latest version: https://travis-ci.org/github/ADVRHumanoids/ROSEndEffector/builds/669781244

torydebra, 2020-04-06:
I found rosinstall and wstool tools. 
wstool permit to create a .rosinstall file like this one : 
~~~yaml
# IT IS UNLIKELY YOU WANT TO EDIT THIS FILE BY HAND,
# UNLESS FOR REMOVING ENTRIES.
# IF YOU WANT TO CHANGE THE ROS ENVIRONMENT VARIABLES
# USE THE rosinstall TOOL INSTEAD.
# IF YOU CHANGE IT, USE rosinstall FOR THE CHANGES TO TAKE EFFECT
- git: {local-name: ROSEndEffector, uri: 'https://github.com/ADVRHumanoids/ROSEndEffector.git',
    version: devel}
- git: {local-name: roboticsgroup_gazebo_plugins, uri: 'https://github.com/roboticsgroup/roboticsgroup_gazebo_plugins.git'}
- git: {local-name: rosee_gazebo_plugins, uri: 'https://github.com/ADVRHumanoids/rosee_gazebo_plugins.git',
    version: devel}
- git: {local-name: rosee_gui, uri: 'https://github.com/ADVRHumanoids/rosee_gui.git',
    version: devel}
- git: {local-name: rosee_msg, uri: 'https://github.com/ADVRHumanoids/rosee_msg.git',
    version: devel}
~~~
So, for fresh installation, a user must have this file in <user_ws>/src/ and then simply call 
~~~bash
rosinstall .
~~~
I am still exploring how to manage branches with this wstool, tell me if you know/like it. 

A problem could also be: where we put this .rosinstall? it should be in the upper folder of this repository

liesrock, 2020-04-06:
I think it would be a good trial! I knew about this but I do not have experience with it: let me know how it works and then we can have a trial also on our travis!

torydebra, 2020-04-06:
wstool is used to create and populate the .rosinstall
For example, to add the main package at devel branch:
~~~bash
wstool set --git ROSEndEffector https://github.com/ADVRHumanoids/ROSEndEffector.git
wstool set ROSEndEffector --version devel
~~~
This must be done for each package. After we must update with:
~~~bash
wstool update
~~~

Now the .rosinstall is ready to use and to be distributed.

liesrock, 2020-04-18:
This is super interesting @torydebra! Can you share with me your .rosinstall so I can try also directly that one?

torydebra, 2020-04-18:
The file contains exactly the things above. I attach here anyway so you can faster copy.
**NOTE** you have to rename it to .rosintall (without extension), github does not permit to upload hidden files neither no extension file. You should put it in the  workspace/src folder, so we can not really put in the repo (it is at one level up). 

Maybe there is the possibility to "run" it in a inner folder, but I see that other packages create a "superRepo" (maybe where the general readme is present) where there is the rosintall file which links all the necessary "real" repository. This is also nice because you can create different superRepo branch with different version of rosinstall so each version will download the right branch of "real" repos.

[rosinstall.txt](https://github.com/ADVRHumanoids/ROSEndEffector/files/4497429/rosinstall.txt)

liesrock, 2020-04-27:
So, I created a package manager and added the instruction here:

https://github.com/ADVRHumanoids/ROSEndEffectorPackageManager

If you have the chance to test it (following the readme), I think we can close this and release the Package Manager public as a main entry point for the framework!

torydebra, 2020-04-27:
Very good :+1: 

Only a few things:
- I would create a "devel" branch there, because in rosinstall we clone from "devels" branch. Then we will have one for master, one for lunar-kinetic (maybe), and so on. This is a method I see in other ros packages.

- From installation instructions, it miss the moveit dependencies. I think moveit is better to install with apt-get so it cannot be put in rosintall file. So we can add this one to the readme. 
  **BUT** I would like to test if `rosdep` works: after `rosinstall` command, `rosdep` should install all external dependencies missing, including moveit and maybe others that we will have in future. I am not sure if this works, I will try it removing temporarly moveit from my system

torydebra, 2020-04-27:
News:

- rosdep does not install moveit. So we do not use it

- I added a "mine" version to install (check [readme](https://github.com/ADVRHumanoids/ROSEndEffectorPackageManager)) because yours does not work. When you clone, a folder with repo name is created, so src is not there.

- Let's use this method of another repo to install. Rosinstall clone repo in the folder where it is, we cannot say him to clone things in the upper directory (so we can not have a rosinstall in ROSEndEffector repository)

liesrock, 2020-04-27:
so we just need an extra:

```
cd ROSEndEffectorPackageManager
```

liesrock, 2020-04-27:
I will add a setup.sh with the dependency installation (for now only moveit) and add the command:

```
./setup.sh
```

before the compilation.

Do you agree on the rest?

torydebra, 2020-04-27:
Ok for the additional `cd` command. But the command `rosinstall src` should be changed to `rosinstall .` , we are already in the folder where the file is. 

Nice also to have a ./setup.sh :+1:

liesrock, 2020-04-28:
setup.sh should in theory install the whole ROS, but we will assume the user has a basic ROS installation on his/her pc, right?

torydebra, 2020-04-28:
Yes, I think if someone is looking for a ROS package (like this one), he obviously knows what ROS is. We can always put a link to the ros website before the code block.

torydebra, 2020-04-28:
As soon we are done with this issue, I will update the documentation with rosinstall instructions

liesrock, 2020-04-28:
Ok nice: let's update the instruction in the main documentation.
I created the devel branch as you suggested on the Manager.
I will close this, but you find any other issues we can reopen it.

closed 2020-04-28 by liesrock