ADVRHumanoids/ROSEndEffector #29: Make UniversalRosee take urdf and srdf file from outside the package

torydebra, 2020-03-25:
Now, we need a <hand>.yaml file in config folder with, for example for schunk:
~~~yaml
ROSEndEffector:
  urdf_path: "urdf/schunk.urdf"
  srdf_path: "srdf/svh-ROSEE.srdf"
~~~

For each new hand, this force us to copy urdf file in the rosee package folder urdf. 
Is there a reason for not using ros parameter file loaded in the launch? Why are we using this yaml file?

liesrock, 2020-04-01:
The idea is to have a set of parameters in the YAML related also with the HAL and not only the URDF and SRDF.

Moreover having a separated YAML file can be an overhead, but it will speed up the integration with the XBot framework.

Anyhow we can think about a ROS parameter solution in parallel.

torydebra, 2020-04-01:
After you explain this, and after XBot discussion, I understood. We can even close this issue. Or maybe use this issue to remember us to integrate a "set_xbot_path" command like xbot do

torydebra, 2020-04-24:
Until now, some part of code (mine) take model from ros param server (set in launch file). Other parts (@liesrock ) use a config file that point to the files. I "merge" the two things. Now all models are taken by that config file. Moveit need urdf and srdf loaded in the server, so UniversalFindActions takes the file from the config and put it in the param server (so it is not anymore the launch file that load them). 
This will also help to change urdf and srdf file position: they do not need to be in our ROSEE folder now, we only need to point the right path in that config file.

@liesrock, see if you like how I did it, I opened a pull request

liesrock, 2020-04-27:
Thanks @torydebra, I merged the pull request.

closed 2020-04-27 by liesrock