ADVRHumanoids/ROSEndEffector #31: [UniversalFindActions] dies when closing

liesrock, 2020-03-31:
Why do we have this behavior for the  UniversalFindActions?

```
roslaunch ros_end_effector findActionsRobotiq_3f.launch

....
....

[UniversalFindActions-2] process has died [pid 18033, exit code -11, cmd /home/lucamuratore/src/ros_end_effector__ws/devel/lib/ros_end_effector/UniversalFindActions __name:=UniversalFindActions __log:=/home/lucamuratore/.ros/log/2f09e1c6-736a-11ea-97a3-00dbdff94f34/UniversalFindActions-2.log].
log file: /home/lucamuratore/.ros/log/2f09e1c6-736a-11ea-97a3-00dbdff94f34/UniversalFindActions-2*.log
```

liesrock, 2020-03-31:
It happened also with test_ee hand.

For Schunk it closes successfully.

torydebra, 2020-04-01:
I am aware of that crash, if you watch the last lines of UniversalFindActions, I build some actions as example. These are built from previous primitives, that with some hands do not exist and so the crash. The primitives are found anyway if you look in the config folder, so you can launch UniversalRosee after. I will add a check anyway

torydebra, 2020-04-01:
It should be ok now 45203ec75fa48da0146092afa87469b4b6b7da10

liesrock, 2020-04-01:
Thanks it looks solved! Probably the final:

```
[ERROR MapActionHandler::getPrimitive] Not found any action 'singleJointMultipleTips_3' with key [ left_hand_Finger_Spread, ] 
```
can be just a warning.

closed 2020-04-01 by liesrock