ADVRHumanoids/ROSEndEffector #33: Change primitive name of PinchStrong and PinchWeak

torydebra, 2020-04-06:
One of the most difficult part in programming: names.
As Nikos suggested, Strong and Weak can make think about forces, and it is not the case. Should we change the names? If so, we need some adjectives to indicate a pinch where the tips collide (so, I called it "strong" because it was a "better" pinch") and do not collide (previously "weak").

torydebra, 2020-04-06:
Some ideas:
**PinchCollision** and **PinchDistance**, which gives also the idea of how these are defined
**Tight** and **Loose**
**Close, Near** and **distant, far, away**

liesrock, 2020-04-18:
Did you find something in the literature on this? I am not super convinced...

torydebra, 2020-04-18:
> Did you find something in the literature on this? I am not super convinced...

No, but I think it is not so easy find something, these subdivision of pinches is a lot specific for our purpose...  I will check anyway

torydebra, 2020-04-20:
In literature, the famous [Cutkosky 1989] the pinch is called :
> precision grasp  – a grasp where the object is held with the tips of fingers

Other works speaks about _palmar pinch_, _tip pinch_  but they are related to the object grasped, while in out work the object is not considered

liesrock, 2020-04-20:
I see: since in the taxonomy (https://www.is.mpg.de/uploads_file/attachment/attachment/256/grasp_taxonomy.pdf) there is only what you reported in the SoA, I would go for what Nikos suggested:

- Tight Pinch
- Loose Pinch

Since we are not referring to the target object to grasp, I think the above is the best choice we have.

torydebra, 2020-04-20:
Changed name in tight and loose in another branch. 
As soon the other pull request #39 is solved, I will merge devel into this branch and force push it into devel

ntsagarakis, 2020-04-20:
Indeed not easy. Even tight and loose do not self explain what we mean but I cannot think anything better.

torydebra, 2020-04-21:
Ahah naming is one of the most diffcult part to do always. 
However, I changed in 5617a56af6213e91078b08d469558422d4039949, I can change again in future if some better names come us in mind.

closed 2020-04-21 by torydebra