ADVRHumanoids/ROSEndEffector #34: A fingertip can also be not a leaf link in urdf

torydebra, 2020-04-08:
An issue that comes out for shadow hand, but can be a problem also for others. 
The shadow in the urdf has as last link for each finger a "virtual" link it has no mesh associated (and also 0 inertia and very low weight). I do not know why there is such a link. So the real tip is the 2nd to last link
Anyway, this causes problems to findActions, because for a pinch, a collision between tips can't be found in this case because there is no mesh.

The solution is to specify the finger (in the urdf) putting as last link of the chain (i.e. the tip) the 2nd to last link of the urdf, so the moveitparser identify this as tip.
Doing this go against my "definition" of fingertips: 
https://github.com/ADVRHumanoids/ROSEndEffector/blob/d554ceca31c05ad58cfcfe6d27582645ba44307e/include/ROSEndEffector/ParserMoveIt.h#L219-L222

So I should relax this definition crossing out the last sentence. I am asking if you think this is ok, and also if defining the finger chain in that way can cause problem to parser and rosee online part done by you @liesrock (I think not)

torydebra, 2020-04-08:
Probably a fake link can be used to attach a sensor, so we should not obly the user to have a mesh as last link

torydebra, 2020-04-09:
I change the method used, now it is:
https://github.com/ADVRHumanoids/ROSEndEffector/blob/7667964d5d6a1bdf726f08067562cfbc0d75cc4c/include/ROSEndEffector/ParserMoveIt.h#L219-L224

liesrock, 2020-04-18:
The solution proposed is ok for me.

closed 2020-04-18 by liesrock