ADVRHumanoids/ROSEndEffector #38: Segfault when trying to send action to Schunk hand

liesrock, 2020-04-19:
Hi @torydebra, probably you are already aware of this issue, but I was trying the supported hands so far and I had the following segmentation fault when trying to send a grasping action to the Schunk hand.

Debugging I found that the issue is here when accessing the jointPos map :

https://github.com/ADVRHumanoids/ROSEndEffector/blob/2d7350ef3a691995cb7ad4e988e471d6cc44b93c/src/UniversalRosEndEffectorExecutor.cpp#L294

I noticed your comment in the line above.

What's your opinion on this?

torydebra, 2020-04-20:
The comment was about a doubt that I had but it really never caused an issue, I only put as suggestion for future possible bugs.

The problem you have is another thing, I recently updated the topic names but I do not update the schunk launch accordingly. I will solve, meanwhile I try also to make possible to launch schunk with generic rosee_startup.launch as the other hands (until now a specific schunk_startup.launch is needed)

torydebra, 2020-04-20:
Ok, done in 9f411889b3240f94b964f8b7075dea21f4e7e01b
Now you can launch schunk as the other hands:
~~~
#source schunk package for meshes
roslaunch ros_end_effector findActions.launch hand_name:=schunk
roslaunch ros_end_effector rosee_startup.launch hand_name:=schunk
~~~
**Note** Please delete _configs/actions/svh_ folder, now schunk action are stored in _configs/actions/schunk_

liesrock, 2020-04-20:
Thanks I will check it asap.

liesrock, 2020-04-22:
I confirmed @torydebra commits solved the issue!

closed 2020-04-22 by liesrock