ADVRHumanoids/ROSEndEffector #46: [Testing] Implement Integration Test

liesrock, 2020-04-30:


torydebra, 2020-05-04:
Probably we need [rostest](http://wiki.ros.org/rostest] for these integration testing

liesrock, 2020-05-10:
http://wiki.ros.org/rostest/Nodes can be the starting point: writing .test files can help to understand the capabilities of the testing framework.

torydebra, 2020-07-28:
Some updates: I have defined a send action tests which check the corectess of sending a created custom action.
The branch is a test_issue40 checkout: we should conclude that pull request ( #44 #40 ) before merging these new updates

torydebra, 2020-07-30:
Interesting for code coverage
https://github.com/ADVRHumanoids/xbot2_wip/issues/21

torydebra, 2020-07-31:
> Interesting for code coverage
> [ADVRHumanoids/xbot2_wip#21](https://github.com/ADVRHumanoids/xbot2_wip/issues/21)

This has a dedicated issue #68

torydebra, 2020-09-17:
> Some updates: I have defined a send action tests which check the corectess of sending a created custom action.
> The branch is a test_issue40 checkout: we should conclude that pull request ( #44 #40 ) before merging these new updates

The updates on integration test (ie test also on sending actions) are in test_issue40 branch now. Also the code coverage is in that branch