ADVRHumanoids/ROSEndEffector #52: Use only one naming convention for package name

torydebra, 2020-05-08:
In some place we use ROSEndEffector, somewhere else we use ros_end_effector. This probably cause dependency issues as suggested by @alaurenzi in issue #48 

From [ROS conventions](http://wiki.ros.org/ROS/Patterns/Conventions) : 

>   Package names should follow common C variable naming conventions: lower case, start with a 
>   letter, use underscore separators, e.g. laser_viewer

So we should change everything to ros_end_effector, removing everywhere ROSEndEffector. 

But in the same web page I see also:

> Prefixing a package name with 'ros' is redundant for a ROS package. This is not recommended except for very core packages.

This can be also important, while we are going toward the 1.0 beta

liesrock, 2020-05-08:
Thanks for the above @torydebra: ros_end_effector should be the name based on the convention.

Regarding the repo name we can leave ROSEndEffector right?

alaurenzi, 2020-05-08:
Yes, I ended up doing the same for CartesianInterface. I'm not totally satisfied with different repo-project name, though..

torydebra, 2020-05-08:
I know that the name ros end effector is already famous :P... but I would think also about removing the *ros_* prefix... I do not know how much are strict the ROS rules but this may cause to do the changing name job twice. 
Then I do not know, maybe ROS does not comply so much.

liesrock, 2020-05-08:
No the ros prefix has to stay there :)

torydebra, 2020-05-08:
ok, so we change everything (but github repo name) to ros_end_effector ?

liesrock, 2020-05-08:
Yes you can go ahead on this! Thanks

torydebra, 2020-05-08:
Done! also in the other packages. I removed the build, devel and compile all from the nothing. I also compile the test and dox (doxygen) to be sure.
There should be no issues

liesrock, 2020-05-10:
I checked it and tested it also looking at Travis: https://travis-ci.org/github/ADVRHumanoids/ROSEndEffector/builds/684669274?utm_source=github_status&utm_medium=notification

I'll close, thanks!

closed 2020-05-10 by liesrock