ADVRHumanoids/ROSEndEffector #53: [Crash] Fail to load the model on param server in time

torydebra, 2020-05-12:
Sometimes it happens that the robot_description parameter is not found by joint_state_publisher and robot_state_publisher. 
~~~ 
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher", line 44, in <module>
    jsp = joint_state_publisher.JointStatePublisher()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/joint_state_publisher/__init__.py", line 145, in __init__
    raise RuntimeError('The robot_description parameter is required and not set.')
RuntimeError: The robot_description parameter is required and not set.
[robot_state_publisher-4] process has died [pid 16835, exit code 255, cmd /opt/ros/kinetic/lib/robot_state_publisher/robot_state_publisher joint_states:=/js_publisher/joint_states __name:=robot_state_publisher __log:=/home/edoardo/.ros/log/70101d06-8b0d-11ea-afc2-0c5415a36088/robot_state_publisher-4.log].
log file: /home/edoardo/.ros/log/70101d06-8b0d-11ea-afc2-0c5415a36088/robot_state_publisher-4*.log
[joint_state_publisher-3] process has died [pid 16827, exit code 1, cmd /opt/ros/kinetic/lib/joint_state_publisher/joint_state_publisher joint_states:=/js_publisher/joint_states __name:=joint_state_publisher __log:=/home/edoardo/.ros/log/70101d06-8b0d-11ea-afc2-0c5415a36088/joint_state_publisher-3.log].
~~~

This seems to happen randomly on my system. It may also be the cause of [this issue](https://github.com/ADVRHumanoids/ROSEndEffectorPackageManager/issues/1).

This problem could be caused because the main rosee node (universalfindaction in offline phase and universalrosendeffector in online phase) does not put the param in time. 
This problem probably happens since commit 4ac21f96fc6e44dccfd042f0d766ae941e17cacf. There, I deleted the param loading in the launch file. This was because somewhere in the code we take urdf file from the ros_ee_config file, and somewhere else directly from param server, so we had 2 points. 
The ideal would be to get the filename from config file, and set the file with this arg all in the launch. But in launch it is not possible to "get" a param. 
To solve we should add again the load in launch file, and keep 2 entry points.

@liesrock, if you have understood all these confabulations, maybe you are aware of a better method to solve.

liesrock, 2020-05-13:
Yes the problem is clear: param server sometimes does not work as expected, in this case the issue is the one you mentioned. 
Since it is clear that we can't use it as a  "dictionary" in the memory, I would go back to the two entry points.

This would solve also @edoardolamon issue.

torydebra, 2020-05-14:
Ok, done. 
In XBot, how is this handled?

closed 2020-07-24 by torydebra