ADVRHumanoids/ROSEndEffector #56: HERI HAND model problem

torydebra, 2020-06-29:
We have discussed about how to use current from real Heri II to understand finger positions (ie joint phalanges). @EvanZhangYifang already did the model for that.
Anyway, we found out that a motor finger, after the finger reached the wanted position, starts to lower the current. @EvanZhangYifang can certanly explain this better than me.

An idea is a method to detect the "hold state" ( where the current begins to fall ) so we can increase some parameter in this state.

Other idea is to brutally increase some params...

EvanZhangYifang, 2020-06-29:
Two states:
Forward transmission state: Motor —> Gearbox (Max. Efficiency: 0.9) —> Finger joints;
Hold state or backward transmission: Finger joints —> Gearbox (Max.Efficiency: 0.9) —> Motor.

In today's experiments, when the finger in close state , from idle to target position, the finger motion in gazebo was very similar with real finger motion. However, when the finger reached the target position, the drive current decreased dramatically, much lower than expect current. This caused the reopen of the finger in gazebo while the real finger hold its position. 

The possible reason is that: The equation that use for convert the current and drive force on tendon , currently, can only present the forward transmission state. After the finger reach the desired position, the finger enter the hold state or backward transmission state, the gearbox was acting as a brake which decrease the brake torque requirement for motor. Furthermore, the maximum transmission efficiency of gearbox represents its efficiency in dynamic condition, when in static condition the factor is much lower. Thus, the real hold current is much lower than the current calculated with Max. Efficiency. 

This factor is unknown. They need to be tested and adjusted.

torydebra, 2020-07-03:
Updates after Nikos meeting. In brief we do not need to monitor the current, we can simply use the motor position feedback

<a href="https://www.codecogs.com/eqnedit.php?latex=F_{tendon}&space;\frac{K_{t}&space;K_{p}&space;(\theta_d&space;-&space;\theta)}{r&space;R_M}&space;=&space;\frac{K_{t}&space;K_{p}&space;(x_d&space;-&space;x)}{r^2&space;R_M}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?F_{tendon}&space;\frac{K_{t}&space;K_{p}&space;(\theta_d&space;-&space;\theta)}{r&space;R_M}&space;=&space;\frac{K_{t}&space;K_{p}&space;(x_d&space;-&space;x)}{r^2&space;R_M}" title="F_{tendon} \frac{K_{t} K_{p} (\theta_d - \theta)}{r R_M} = \frac{K_{t} K_{p} (x_d - x)}{r^2 R_M}" /></a>