ADVRHumanoids/ROSEndEffector #65: Use srdf tag group state to let user says the default joint pos

torydebra, 2020-07-29:


torydebra, 2020-08-04:
Some problems for this, I checkout at new branch https://github.com/ADVRHumanoids/ROSEndEffector/tree/srdfInitialPos

torydebra, 2020-08-05:
ROSEE is ready to use with this new feature, I have also tested it a bit.

Problem is that the joint_state_publisher does not read srdf, thus it is not aware of possible initial joint position different from 0.
Possible solutions are:
- Use a yaml file where the _zeros_ of the joint are written. This is the standard way documented for joint state pub. The problem here is to have this additional duplicated file, and we need it for each end-effector, even if the joint position are all 0.
- Modify joint_state_pub such that it reads srdf
- Send a init command from ROSEE after parsing the robot with its initial joints positions. Not really coheret though, ROSEE should always pass though the filter and the hal to send robot commands.
- Send a init command from a "initialization ros node", then shutdown it and run rosee.

Similar problem happens also when gazebo (and not anymore joint state pub) is used, but it should be easier to set initial pos in gazebo

torydebra, 2020-08-05:
BTW, I think this feature is not so important, so we can delay its integration

lia2790, 2020-08-05:
I think y can leave it...