ADVRHumanoids/ROSEndEffector #70: [TRAVIS] travis.org migration into travis.com

torydebra, 2020-09-16:
As I read in a recent [email](https://mailchi.mp/3d439eeb1098/travis-ciorg-is-moving-to-travis-cicom?e=e645c16fb4), travis is closing down the .org version. We are using .org for the rosee public repos.

@liesrock, should I proceed to migrating to .com, as explained [here](https://docs.travis-ci.com/user/migrate/open-source-repository-migration) ?

liesrock, 2020-09-16:
Thanks @torydebra , yes this would be great: if you need permission (we are paying the .com version for private repos) let me know.

torydebra, 2020-09-16:
Tagging everyone, this could be interesting also for other advr repos @ADVRHumanoids/advr-developers

liesrock, 2020-09-16:
Thanks again @torydebra, I was able to perform the migration: in case somebody needs it for the future, you should sign up for the beta to be able to migrate .org travis CI to .com

https://travis-ci.com/organizations/ADVRHumanoids/repositories

closed 2020-09-16 by liesrock